victoriarobotics/victoria_electrical #4: Build Teensy Interface Board

griswaldbrooks, 2017-01-30:
Need board that the Teensy and all the things the teensy needs can plug into.
Will also have the 5V rail for computer.

griswaldbrooks, 2017-01-30:
@griswaldbrooks Need to add connector part numbers

griswaldbrooks, 2017-01-30:
Here are the vertical locking connectors for the encoders, power, and serial.

- 4 Pin
  - WM4802-ND
  - http://www.digikey.com/product-detail/en/molex-llc/0705430003/WM4802-ND/114923

- 3 Pin
  - WM4801-ND
  - http://www.digikey.com/product-detail/en/molex-llc/0705430002/WM4801-ND/114919

- 2 Pin
  - WM4800-ND
  - http://www.digikey.com/product-detail/en/molex-llc/0705430001/WM4800-ND/114915

griswaldbrooks, 2017-01-30:
Female headers for Teensy
- S7057-ND
- http://www.digikey.com/product-detail/en/sullins-connector-solutions/PPPC241LFBN-RC/S7057-ND/810196

markwomack, 2017-02-13:
Circuit is soldered according to current requirements.  Room for future growth.

[image]
[image]
[image]

markwomack, 2017-03-02:
The base board is built.  I am closing this one and creating bugs for remaining, specific issues.

closed 2017-03-02 by markwomack