victoriarobotics/victoria_electrical #5: Install New Mains Power Switch

griswaldbrooks, 2017-01-30:


griswaldbrooks, 2017-02-05:
Installed.

closed 2017-02-05 by griswaldbrooks

griswaldbrooks, 2017-02-05:
Switch from the outside.
[image]

New switch can be seen installed in the lower left hand corner.
[image]

griswaldbrooks, 2017-02-05:
The switch was put in by opening up a hole a little bit in the existing rear port. As the original hole was square, there were some gaps which were closed by installing a washer and o-ring on the inside (non-weather facing) of the robot. The boot nut was tightened slightly, but the cramped space made it difficult to tighten fully.

ralphhipps, 2017-02-05:
looks good!!!

Ralph

>+<
 |    /\
 |___/  \
 [_@_]   `<
 O O O          IRS!
------------------------------------------------------------------------------------
In the twenty-first century, the robot will take the place
which slave labor occupied in ancient civilization.
 -- Nikola Tesla
------------------------------------------------------------------------------------