victoriarobotics/victoria_electrical #15: Wire runstop

griswaldbrooks, 2017-03-06:
Runstop needs to be wired into system.
The first choice was to wire it into the Teensy, but given that the Teensy has had issues, should we consider wiring it directly to the motorcontroller?

@markwomack

griswaldbrooks, 2017-03-09:
Attached cable to runstop.
Three wires, black, white, and red.
Made black common to both the NO and NC sides, and wired the other red/white wire to the other side (don't remember which).
This way, we have the option of sensing NO or NC.

griswaldbrooks, 2017-03-20:
Going to investigate the use of Ch 5 on the TRex from #17. If this works, then a small circuit will be in order to take the e-stop as well as a deadman, and convert it to a PWM for the TRex.
It should also tell the Teensy that something has stopped the robot, so it can send a cmd_vel=0 command as the TRex will/might try to execute the last serial command when un-e-stopped, assuming the Teensy is alive.

ralphhipps, 2017-03-20:
Can the Teensy do that, or too much added latency? Burden on the processor?

I assume it can generate PWM easily, so read e-stop & deadman, generate PWM
when needed.

Thots?

> Going to investigate the use of Ch 5 on the TRex from #17
> <https://github.com/victoriarobotics/victoria_electrical/issues/17>. If
> this works, then a small circuit will be in order to take the e-stop [...]

-- 
Ralph

>+<
 |    /\
 |___/  \
 [_@_]   `<
 O O O          IRS!
------------------------------------------------------------------------------------
In the twenty-first century, the robot will take the place
which slave labor occupied in ancient civilization.
 -- Nikola Tesla
------------------------------------------------------------------------------------

griswaldbrooks, 2017-03-20:
The Teensy could handle this without a problem wrt processor load or io.
The issue is the Teensy power has been unstable lately (hopefully fixed) and the Teensy code is under development. Sometimes, for some reason this causes the processor to stop working and the TRex continues to run the last command it received. Then I have to catch the robot and physically lift it up, which is not ideal.

griswaldbrooks, 2017-03-20:
I've got plenty of extra arduino-esque boards I can use to make the runstop/deadman circuit.

griswaldbrooks, 2017-03-20:
Also need to add a deadman disable jumper.

griswaldbrooks, 2017-03-20:
I have these in mind for the deadman so we don't have to carry a full RC remote:
https://www.adafruit.com/product/1096
https://www.adafruit.com/products/1095

The keyfob I'd have to hack up so it was easier to hold.

griswaldbrooks, 2017-04-07:
@markwomack how do you feel about the Teensy stability at this point. Should we rely on it for our deadman? I'm going to order those switches I listed so I have have them soon.

griswaldbrooks, 2017-04-10:
@markwomack wired this into the Teensy, with a pull down. When the button, isn't pressed, it applies a voltage to pull up the pin.

closed 2017-04-10 by griswaldbrooks